robofit/arcor2 #75: Pivot point for aiming objects

Kapim, 2020-04-15:
See https://github.com/robofit/arcor2_editor/issues/58

Current pivot point is in the center of the object, which makes imposible to aim existing object. We should change this - at first to center of top plane, later it would be nice that user may choose the pivot point (e.g. from list like center of top plane, center of front plane etc.), to allow aim high objects or walls for example

ZdenekM, 2020-04-16:
@Kapim Ok. Moreover, it would be nice if AR Editor could visualize model of the collision object attached to the robot's end effector - so an operator could overlay model with the real object. Without such visualization, the feature will be probably hard to use.

rstourac, 2020-04-16:
Example from real world: We have simplified ICT tester on the table. Its dimensions are 180x180x80mm. It is not big problem to navigate robot in the center of a object in X and Y axis. Z was aimed 20mm above the object. I created collision object 200x200x100mm. If I hold object in the middle of top plane, it would be positioned as expected. I think this would be enough for MVP version.

ZdenekM, 2020-04-22:
@rstourac Sure, I understand. Functionality is already prepared in ARServer, it will probably appear in a next version of AREditor.

closed 2020-04-22 by ZdenekM