robofit/arcor2 #85: Improve work with execution packages

ZdenekM, 2020-04-21:
- [ ] Read scene/project from running execution package and forward it to Editor.
- [ ] Rename RunProject etc. to RunPackage.
- [ ] Add "project_id" to ListPackagesResponse/PackageSummary.
- [ ] Enable work on project while execution package runs (only if the scene is same?), just disable action execution.

Kapim, 2020-04-23:
- [ ] Add name to ListPackagesResponse/PackageSummary
- [ ] Add name to BuildPackage

ZdenekM, 2020-04-30:
Work on project while package runs could be enabled in the future - it has to be decided what should and what should not be possible to do.

ZdenekM, 2020-05-08:
@Kapim Maybe, we might merge SceneCollisions into PackageInfo? Original idea was to have it separately in order to allow dynamic updates of collision models but it is not needed now and maybe it complicates things a bit. What do you think? Anyway, collision models of the objects should not change during script runtime.

Kapim, 2020-05-08:
I am definitely all for it. I have not implemented this yet and the merge will simplify it for me.

ZdenekM, 2020-05-11:
Ok. Done in https://github.com/robofit/arcor2/commit/4e6c6753117d0b855633ec9588bc90b04e7b3fd2.

closed 2020-05-26 by ZdenekM