robofit/arcor2 #108: Action id not found in object

NelaMm, 2020-05-08:
After save project and click Run project
[2020-07-5--16-55-40.zip](https://github.com/robofit/arcor2_editor/files/4593735/2020-07-5--16-55-40.zip)

ZdenekM, 2020-05-08:
@NelaMm This is not issue of AREditor, so I transferred it here. It is really strange - it could not find action "move" which actually IS in the project. Could you please attach archive with the execution package?

NelaMm, 2020-05-14:
This happen each time in new AREditor 0.5.2
Execution package: [32e354e3af3c4e73b6a3586771bae8b1.zip](https://github.com/robofit/arcor2/files/4628807/32e354e3af3c4e73b6a3586771bae8b1.2.zip)

ZdenekM, 2020-05-15:
@NelaMm  Based on the traceback found in the package you uploaded, I think you have current version of ARServer (0.4.2), but outdated version of Execution service (0.3.0). Can you check version of the execution service? E.g. using ```arcor2_execution --version``` within the container or check version in its swagger-ui (http://execution:5009/swagger/).

ZdenekM, 2020-05-15:
We will add some further checks (see issue https://github.com/robofit/arcor2/issues/119) in order to prevent problems like this.

ZdenekM, 2020-05-22:
> @NelaMm Based on the traceback found in the package you uploaded, I think you have current version of ARServer (0.4.2), but outdated version of Execution service (0.3.0). Can you check version of [...]

@NelaMm Could you confirm my suspicion? Is the issue resolved?

NelaMm, 2020-05-22:
Yes, I use version 0.3.0.

ZdenekM, 2020-05-22:
@NelaMm Ok. So please make sure to use exactly the same versions of arcor2_arserver, arcor2_build and arcor2_execution. Otherwise errors like this might occur.

closed 2020-05-22 by ZdenekM

NelaMm, 2020-05-22:
Ok, thanks for information.