robofit/arcor2 #234: Arcor execution service fails to list the uploaded packages

Croolman, 2020-09-07:
When an attempt is made to list the uploaded packages an exception is thrown.

Steps to reproduce:
1. Create a .zip file containing only the .py script.
2. Upload the file.
3. Get the packages.
4. Exception is thrown. Following is the traceback.
```
Error in connection handler
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/websockets/server.py", line 191, in handler
    await self.ws_handler(self, path)
  File "/root/arcor2/arcor2/helpers.py", line 183, in server
    resp = await rpc_dict[req_cls](req, client)
  File "/root/arcor2/arcor2/nodes/execution.py", line 302, in list_packages_cb
    with open(os.path.join(folder_path, "data", "project.json")) as project_file:
FileNotFoundError: [Errno 2] No such file or directory: '/root/project/tt2/data/project.json'
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/root/arcor2/arcor2/nodes/execution_proxy.py", line 110, in ws_thread
    data = json.loads(ws.recv())
  File "/usr/local/lib/python3.8/json/__init__.py", line 357, in loads
    return _default_decoder.decode(s)
  File "/usr/local/lib/python3.8/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/lib/python3.8/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
```
Note: The execution of the package works as intended.

ZdenekM, 2020-09-07:
@Croolman Could you please check what version of the Execution service are you using? From the traceback, it seems like 0.7.1 which could not work without scene/project.json in the package. You should be using 0.8.0b* (where 0.8.0b8 is the latest).

Croolman, 2020-09-07:
@ZdenekM Sorry, forgot to mention I am using the 0.7.1 version. I have tried the others, but those were not working as well. The misconception was, that we should be using the docker images from `arcor2/arcor2_execution_kinali` located on docker hub. The docker from `arcor2/arcor2_execution` with the tag you are mentioning is working fine (at least for `hello world` type of script). 

Closing as solved. Thank you.

closed 2020-09-07 by Croolman