robofit/arcor2 #240: Timeout on interaction service

hanapelikanova, 2020-09-14:
Default timeout on Interaction service is too short. In order to have enough time to resolve dialog on Interaction service, it is necessary to extend timeout up to 8 hrs.

ZdenekM, 2020-09-15:
We are going to rework the rest module soon. Each method will probably get parameters to specify a timeout. Is it ok to wait for it until the 0.9.0 version (0.8.0rc1 will be released today, 0.8.0 probably this or the next week) or do you need any temporary solution?

martincervinka, 2020-09-15:
Change it ASAP, please.

ZdenekM, 2020-09-15:
As I wrote above - the module is going to be rewritten (at the latest in 0.9.0, we can try to make it sooner). Unfortunately, it is quite a complex task and we are not able to do it immediately. Right now, the module only allows setting timeout globally, for all requests. So in the meantime, please set the timeout in the main script (see https://github.com/robofit/arcor2/pull/244 for an example).

martincervinka, 2020-09-15:
Is there only one REST module for all clients (robot, search, ...)?

ZdenekM, 2020-09-15:
There is only one REST module (part of arcor2), shared by all modules/services that communicate over HTTP/REST.

martincervinka, 2020-09-15:
I mean instances. Can I have more instances of REST client with different timeouts? E.g. Robot - 1 minute, Interaction - 8 hours, Statistic - 1 second.

ZdenekM, 2020-09-15:
So far, this is not possible. That is one of the reasons why we are going to rewrite the module.

ZdenekM, 2020-10-08:
The rest module was rewritten (https://github.com/robofit/arcor2/pull/293). Now, it is possible to specify the timeout for each call to REST API. All 'move' actions of the robot have [prolonged timeout](https://github.com/robofit/arcor2/blob/master/src/python/arcor2_kinali/object_types/abstract_robot.py#L3) as well as call to `add_dialog` of the `Interaction` service ([link to code](https://github.com/robofit/arcor2/blob/master/src/python/arcor2_kinali/object_types/interaction.py#L57)). Let me know if you need custom timeouts somewhere else or open the PR with changes. Btw, the [default timeout](https://github.com/robofit/arcor2/blob/master/src/python/arcor2/rest.py#L45) is 3.05s for connection and 20s to get a reply (that is probably quite a lot). You can even change the default value from the main script (monkey-patch the `Timeout` named tuple).

closed 2020-10-08 by ZdenekM