robofit/arcor2 #253: Execution service fail on empty scene_id in project.json

NelaMm, 2020-09-18:
Execution service fail on upload project contain file project.json with scene id="".

File with logs from execution and executin proxy service:
[log.txt](https://github.com/robofit/arcor2/files/5244771/log.txt)

ZdenekM, 2020-09-21:
> Execution service fail on upload project contain file project.json with scene_id="" or some id (when scene.json not exist).
I just tried to upload a package where project.json contained scene_id="" and it went fine... If the issue persists, could you maybe attach an example package where this happens? But anyway, I would not probably consider this as a bug - if there is a project.json, there should also be scene.json. It now works (well, at least for me :-)) just by chance.

> Fail because of project.json not contain property "modified".
This can be fixed easily...

NelaMm, 2020-09-21:
> I just tried to upload a package where project.json contained scene_id="" and it went fine... 

Sorry, my description was bad. I meant I filled only required parameter (id, name and scene_id) and it didn´t work. So the value of scene_id was relevant.

ZdenekM, 2020-09-25:
So the Execution service now does not require ```modified``` property in the ```project.json```. Is this issue solved now or is there something else that has to be fixed?

NelaMm, 2020-09-29:
Yes, it's solved.

closed 2020-09-29 by ZdenekM