robofit/arcor2 #275: Not able to set parameter Pose for service's method set configuration

NelaMm, 2020-09-29:
arcor2_kinali version 0.8.0

KinaliAbstractObject need parameter Pose in constructor.
Method system/set (eg. service Search, Barcode) have optional body parameter Pose, but not possible to set.

ZdenekM, 2020-09-29:
This is intentional - but maybe it is a misconception on our side. Search / Barcode used to be services, now they are objects without pose. From our point of view, currently, it only makes sense to set pose to a robot. Other objects/services are somehow virtual. I think there is an idea that if the user wants visualization of e.g. Search's frustum in the scene, there will be some child object...

NelaMm, 2020-09-29:
For barcode it is not necessary but rest call system/set still contain this optional parameter. For search it is needed becase of relation between camera and world(robot). For example if we use method compute_pick_args.

ZdenekM, 2020-09-29:
> For barcode it is not necessary but rest call system/set still contain this optional parameter.

As I'm thinking about it, it might make sense to set pose for Barcode - imagine a situation with more readers. Then the user might set the position of the object so it corresponds with the real reader. Otherwise, it might be difficult to figure out which reader is which one.

> For search it is needed becase of relation between camera and world(robot). For example if we use method compute_pick_args.

Ok, no problem. So the pose of `Search` is actually the pose of the camera, right?

Now, the `Search` class is based on `KinaliAbstractObject` which is based on `Generic` - the base object that has no pose. We could prepare `KinaliAbstractObjectWithPose` based on `GenericWithPose` (a class that adds pose to `Generic`) and `Search` would be derived from it - so it will be possible to set its pose. Or more precisely - it will be necessary to set its pose. We do not support an optional pose. Objects either have a pose or do not (based on their base class).

I'm proposing this state.... Objects with a pose (based on `GenericWithPose):
 - `Barcode`
 - `Robot`
 - `Search`

Objects without pose (based on `Generic`):
 - `Interaction`
- `Statistic`

Is it ok like this?

...or if you demand to be able to set pose to all Kinali objects (including `Interaction` and `Statistic`), we can simply change the base of `KinaliAbstractObject` from `Generic` to `GenericWithPose`. But it will make work with AREditor slightly confusing in my opinion, as 3D would contain objects that have no counterpart in the real world.

NelaMm, 2020-09-30:
> Otherwise, it might be difficult to figure out which reader is which one.

Each reader have own config id and id and web api url, so after start scene with reader should be manipulated on its own web API.

> Ok, no problem. So the pose of `Search` is actually the pose of the camera, right?
Yes that's right.

> I'm proposing this state.... Objects with a pose (based on `GenericWithPose):
> 
> * `Barcode`
> * `Robot`
> * `Search`
> 
> Objects without pose (based on `Generic`):
> 
> * `Interaction`
> * `Statistic`
> 
> Is it ok like this?

It looks OK. I agree with you. 

>...or if you demand to be able to set pose to all Kinali objects (including Interaction and Statistic), we can simply change the base of KinaliAbstractObject from Generic to GenericWithPose. 

We do not configure this services, so it makes no sense to us.

ZdenekM, 2020-09-30:
@NelaMm Implemented in https://github.com/robofit/arcor2/pull/280. There are three abstract object types now, some names were simplified, the example script updated... Could you briefly check if it is ok like this?

PS: I've sent you an invitation to the [ARCOR2 team](https://github.com/orgs/robofit/teams/arcor2) so I can assign you as a reviewer for this kind of pull requests.

NelaMm, 2020-09-30:
Thanks. It look good, but I think that is missing cleanup method that call system/reset in class AbstractWithPose.

ZdenekM, 2020-09-30:
Fixed.

closed 2020-09-30 by ZdenekM