robofit/arcor2 #338: Pause package not work

NelaMm, 2020-10-27:
Execution service proxy 0.8.2 use execution service 0.9.0

PUT /packages/pause not interrupt script execution.

ZdenekM, 2020-10-27:
How does your script look like? Is it hand-written? If you do not use the generated script, there is probably no `Resources` context manager that handles some things for you. In the handwritten script, you have to call `patch_object_actions` with each ObjectType as a parameter, if you want to be able to pause script execution.

Example:
```python
from arcor2.action import patch_object_actions
from arcor2_kinali.object_types.kinali_robot import KinaliRobot
patch_object_actions(KinaliRobot)  # this adds action decorator to all actions of the ObjectType
```

...as a side effect, when you run such script manually (not within the Execution unit) it will be printing a lot of JSONs.

NelaMm, 2020-10-29:
I did not realize that. Thank you for explanation.

ZdenekM, 2020-10-29:
It wasn't documented anywhere... It is now in the [example script](https://github.com/robofit/arcor2/blob/master/src/python/arcor2_kinali/scripts/example_main.py) (added in https://github.com/robofit/arcor2/pull/340).

closed 2020-10-29 by ZdenekM