robofit/arcor2 #341: AR Editor not work with kinali object types

NelaMm, 2020-10-29:
**Description**
New AR Editor version 0.10.1 doesn't work after upload kinali services. Arserver crashes after upload these services (added log below). 

**Found in version**
AREditor version 0.10.1
arcor2_arserver 0.10.0
arcor2_upload_kinali 0.9.1

**Perform the following steps reproduce the bug**
- Run docker by docker-compose
- Run arcor2_upload_kinali for uploading services
- Run AREditor

**Observed results**
AREditor can not connect.

**Expected results**
AREditor succesfully connected and objects succesfully added.

**Related data**
[log.txt](https://github.com/robofit/arcor2/files/5457484/log.txt)

ZdenekM, 2020-10-29:
Thanks for the report. We will release fixed images ASAP.

Just a note (you probably know, but it is not mentioned in your steps to reproduce): after uploading object types (running arcor2_upload_kinali) you have to restart ARServer - for now, it downloads and analyzes object types only on start-up.

ZdenekM, 2020-10-29:
Btw, if you want to continue with the testing, you can update [KinaliRobot](https://github.com/robofit/arcor2/blob/master/src/python/arcor2_kinali/object_types/kinali_robot.py)'s sources in the Project service.  The problem was there and is already fixed. At the same time, error handling in the ARServer was improved, as well as testing so this should not happen again hopefully. @Kapim will release new versions of arcor2_arserver and arcor2_upload_kinali today or tomorrow in the morning.

NelaMm, 2020-10-29:
Ok. Thank you.

ZdenekM, 2020-11-02:
Please try now with arcor2_arserver 0.10.1, arcor2_build 0.9.2 and arcor2_kinali 0.9.2. The root cause lied in one transitive dependency - we updated the library version and the problem seems to be gone.

NelaMm, 2020-11-02:
Seems to be OK.

ZdenekM, 2020-11-02:
Glad to hear that! :-)

closed 2020-11-02 by ZdenekM