robofit/arcor2 #444: Execution service output

martincervinka, 2021-02-12:
Execution service floods logs with unimportant messages. Can we turn off logging or change the priority for GET methods? 
`execution-proxy                        | 10.1.0.240 - - [12/Feb/2021 08:58:01] "[37mGET /packages/executioninfo HTTP/1.1[0m" 200 `
`execution-proxy                        | 10.1.0.240 - - [12/Feb/2021 08:58:01] "[37mGET /tokens HTTP/1.1[0m" 200 -`
`execution-proxy                        | 10.1.0.240 - - [12/Feb/2021 08:58:01] "[37mGET /packages HTTP/1.1[0m" 200 -`

The second problem is that all prints from scripts are logged as errors. 
`print("Služby inicializovány.") -> arcor-execution                        | 2021-02-12 08:57:15 - ERROR   : Služby inicializovány.`

ZdenekM, 2021-02-12:
Currently, it is not possible to change the logging level. We will probably simply set the logging level to ERROR, so this message about API calls will be gone. Or do you need to be able to set it?

Regarding the second problem - the script communicates with the Execution unit through the printing of JSONs containing its execution state. When the script prints something which is not JSON, it is printed out as an error because it is an abnormal state, the text could be possibly traceback and that's why it is logged as an error. Certainly, we can improve this. We just didn't expect there will be calls to `print` in the script. Would you prefer to log it as INFO, or not show it at all? Or, what about using JSON-printing logger (which output can be gracefully handled by Execution) within the script? Or what about preparing the Logging service with `log` action?

martincervinka, 2021-02-12:
I see, ... Can we divide it into two separate issues?

> We will probably simply set the logging level to ERROR, so this message about API calls will be gone.

Yes, should be enough for now and quite easy. Can you do that?

We will discuss more how to handle logging from the script and probably create new issues in the future.

ZdenekM, 2021-02-12:
> Yes, should be enough for now and quite easy. Can you do that?

Sure. Let's keep this issue for output of execution-proxy - I will close it once this is done. When needed, please open a separate issue for logging within the script.

ZdenekM, 2021-02-12:
Implemented in https://github.com/robofit/arcor2/pull/446.

closed 2021-02-12 by ZdenekM