robofit/arcor2 #512: Check if collision exist before deleting

NelaMm, 2021-03-29:
Please control in method  'delete_collision_id' if collision exist before deleting.

https://github.com/robofit/arcor2/blob/4925dd4206e0e91971e71e077cdc8eebe644c58e/src/python/arcor2/clients/scene_service.py#L73

Somewhere in classes cleanups is removing of collision, but  sometimes it already isn't in scene.

ZdenekM, 2021-03-29:
Do you have an idea how it happens that the collision is already gone when the object tries to clean it up?

NelaMm, 2021-03-29:
For example when object is removed by script and after that is destroy of scene that call deleting or may be in method object.cleanup() and when scene is cleanup and after then on destroy object whis is already removed from scene.

ZdenekM, 2021-04-12:
I think that the `delete_colission_id` should fail (raise an exception) if the collision does not exist. In my opinion, it would be better to handle the issue in the `cleanup` method of `GenericWithPose` - first check if the collision model exists and if so, then delete it (or just try to delete it and ignore the potential error). Or we might prepare some convenient methods for `GenericWithPose` like `disable()` and `enable()` which would delete/re-create the collision model - that might even help you simplify the code a bit.

NelaMm, 2021-04-12:
I agree with handle the issue in the `cleanup` method. Adding new methods to `GenericWithPose` is not necessary, but it would be nice.

ZdenekM, 2021-04-13:
Should be fixed by https://github.com/robofit/arcor2/pull/520/commits/3b8a89ae728702ea5dccc3cb07c8938cd4f5dff6. Now, you do not need to manipulate the collision model of the object manually and you can only set `enabled` property to `True` or `False`. Let me know if this does not suit your needs :-)

closed 2021-04-13 by ZdenekM