robofit/arcor2 #567: Improve imported package validation

ZdenekM, 2021-05-13:
When importing an execution package, the Build service currently, for each object type in scene.json, tries to:
- Parse its code (to check if the syntax is valid Python).
- Import the ObjectType's class definition (to check whether all imports work and all ancestors are available).

Additionally, it should:
- Raise an error if `ObjectType.abstract()` is `True` (abstract type can't be used in a scene as it has undefined/abstract methods and/or is marked as abstract manually/intentionally).
- ??

martincervinka, 2022-01-05:
Is this issue somehow related to https://github.com/robofit/arcor2/issues/690? @ZdenekM, can you explain it, please?

ZdenekM, 2022-01-05:
@martincervinka It is a duplicate, I will close it. But that brought me to the question - should we retain the import functionality? Do you use it? Let me know and I will update the issue.

closed 2022-01-05 by ZdenekM