robofit/arcor2 #583: Allow partial import of an execution package

ZdenekM, 2021-05-25:
Currently, the Build unit's method `PUT /project/import` imports the whole package (scene, project, Object Types, and their models) or nothing.

[image]

There should be flags allowing to control what will be imported:
- `importScene`
  - If `importScene and not importObjectTypes`, all types referenced by the scene being imported must already exist on the Project service.
- `importProject`
  - if `importProject and not importScene`, the scene must already exist on the Project service.
- `importObjectTypes`

By default, all of these flags will be set to `true`. If someone wants to import e.g. only Object Types, `importScene` and `importProject` flags have to be set to `false`.

martincervinka, 2021-06-02:
With flags to control what will be imported, the overwrites for each entity (Scene, Project, ...) is not necessary. There should be only one `overwrite` flag. In case there is any entity with the same ID on the Project service, that flag must be set to `true` for processing the import and all entities with the same ID will be overwritten, otherwise, the request fails with an error that describes what type of entities need to be overwritten. The data model for the error will be specified.
- `importObjectTypes` flag automatically handles the import of `collisionModels`, that are referenced from imported `objectTypes`.
- `importProject` flag automatically handles the import of `projectSources` if the project is defined without logic.

martincervinka, 2021-07-22:
KINALI will use methods on Project service for import and export data to and from the TIO system. There are no changes required. The issue can be closed.

ZdenekM, 2021-07-22:
So maybe `PUT /project/import` can be removed?

closed 2021-07-22 by ZdenekM