robofit/arcor2 #606: Project without logic: generate Actions class

ZdenekM, 2021-06-16:
Currently, for a project without logic, actions can be specified within the project, but are ignored during the build process. The intended state is to generate a class for them, which can be used during manual programming within the main script. For a project with logic, the main script currently looks like this:

```python
def main(res: Resources) -> None:
    aps = ActionPoints(res)
    aubo: Aubo = res.objects['a05cf574591442109d5636bd8312026b']
    random_actions: RandomActions = res.objects['d8a5065176d679c1c7cd5aee6984955a']
    time_actions: TimeActions = res.objects['1c7cd5aee6984955ad8a5065176d679c']
    while True:
        aubo.move(
            'EE_Gripper',
            aps.global_ap.poses.default,
            MoveTypeEnum.LINE,
            0.5,
            0.5,
            True,
            an='move')
        rand_interval = random_actions.random_double(0, 1, an='generate_random_double')
        time_actions.sleep(rand_interval, an='sleep')  # example of how action can use results of some previous action

if (__name__ == '__main__'):
    try:
        with Resources() as res:
            main(res)
    except Exception as e:
        print_exception(e)
```

For a project without logic, there probably won't be support for the `link` parameter type (usage of previous results). Then, the main script could look like this:

```python
def main(res: Resources) -> None:
    actions = Actions(res)

    while True:
        actions.move()
        time_actions.sleep()

if (__name__ == '__main__'):
    try:
        with Resources() as res:
            main(res)
    except Exception as e:
        print_exception(e)
```

And the auto-generated `Action` class:

```python
class Actions:
    def __init__(res: Resources) -> None:
         self.aps = ActionPoints(res)

    def move() -> None:
        self.res.objects['a05cf574591442109d5636bd8312026b'].move(
            'EE_Gripper',
            self.aps.global_ap.poses.default,
            MoveTypeEnum.LINE,
            0.5,
            0.5,
            True,
            an='move')

    def sleep() -> None:
        self.res.objects['1c7cd5aee6984955ad8a5065176d679c'].sleep(1, an='sleep')
```

ZdenekM, 2021-06-22:
@martincervinka Is the proposed form of `Actions` class ok for you? Any further requirements or comments?

martincervinka, 2021-06-29:
> For a project without logic, there probably won't be support for the link parameter type (usage of previous results).

Without the ability to use the previous result, the Action class is quite pointless for Python developers. In general, scripting should be for difficult projects that can't be completely created in AREditor, and these ones will require that.