robofit/arcor2 #681: Visualization of Action points in Debug

martincervinka, 2021-11-03:
As an advanced operator using Debug view in WebUI, I would like to see the current ActionPoint highlighted, so that I know in which part the program is located.

- Add property to ExecutionInfo data model.
  - ActionPointId: string - The ID of Action point where currently executed action belongs.

If the requirement is too difficult to implement. value can be defined only for action points that are breakpoints for the first version.

ZdenekM, 2021-11-04:
"The ID of Action point where currently executed action belongs." should probably rather be "The ID of Action point which pose or joints are used as a parameter in currently executed action"? Action point can belong (at least in my mental model :-)) to action only in case when it is defined in `project.json`, but in your case, actions are defined only in code, so the only linkage to the action point is when AP's pose/joints are used as an argument, right? Btw, there might be more poses/joints as an argument to the action, e.g.: `move_trough_waypoints(pose1, pose2, pose3)`. An alternative approach might be to add:

- PosesJointsIds: list[string] - The list of IDs of orientations and joints used as an argument to the currently executed action.

...in ARCOR2, all IDs are globally unique so there can't happen collision and from ID you can always determine what kind of "object" it is. The field would be missing for projects with logic defined in `project.json` (does not make sense).

martincervinka, 2021-11-11:
Yes, that's a more precise definition. We are fine with the current solution.

ChrnyaevEK, 2022-01-10:
There are several solutions to extract AP ID, one of them is to modify Execution rest proxy service so it returns AP ID based on executed Action ID. This is suitable when Actions are bounded to the AP straight away, how ever, when we write script manually there is no straight reference from AP to all its Actions. So this solution is not currently suitable for Kinali.

Another solution is to raise AP ID upstream from the Execution service (inside `arcor2_runtime.action.action` decorator) to Execution rest proxy service. When action is called from script we will dispatch an event with AP ID. The problem is, that I can see that the action decorator require Action name to ID mapping (`ACTION_NAME_ID_MAPPING_ATTR`), which again leads us to direct binding between AP and Actions. To overcome this we could define another event, which will be fired as soon as any AP is met or somehow pass this mapping stage and dispatch `ActionStateBefore` with empty Action ID. 

Are there any other solutions we did not notice? What would be your suggestions?

ZdenekM, 2022-01-11:
You are right... Well, another solution (I think we discussed it a bit and @martincervinka said that it will probably be enough at the time) would be to extend the `PackageState` event and send AP ID together with `PAUSED` state - when the package execution stops on a breakpoint.

https://github.com/robofit/arcor2/blob/6dab12458e1081407989f61b34da71b9840be12d/src/python/arcor2_runtime/action.py#L94

...if this is not sufficient, we may discuss whether it is better to add a new event or extend/modify `ActionStateBefore` to better fit with projects without logic and even without actions. I would probably vote for adding a new event, that will only be sent for projects without logic, maybe even only when a certain flag is set?

@Kapim Is the current data in `ActionStateBefore` ok for visualization of a running program? Or do you see any space to make the situation in the editor easier? Thanks!

martincervinka, 2022-01-11:
Yes, it's sufficient for now, but I don't see a problem implementing a complete solution already. Not even more difficult. Of course, I don't know the details.

I would vote for extending ActionStateBefore with AactioPointId property. AactioPointId quite belongs to that event. More events can be confusing especially with `if` for projects logic.

All the parameters should be optional and the clients should handle the case where any of the parameters are missing. Even AREditor/Server could use the ActionPointId property directly.

ZdenekM, 2022-01-13:
Ok then. We discussed it with @Kapim and agree to change `ActionStateBefore` somehow like this:

```python
@dataclass
class ActionStateBefore(Event):
    @dataclass
    class Data(JsonSchemaMixin):

        # required for projects with logic (where actions are defined in the project)
        # might be also filled for projects without logic, if there are actions defined
        action_id: Optional[str] = None  # id of the action that is going to be executed
        parameters: Optional[List[str]] = None  # runtime values of action parameters

        # required for projects without logic and without actions defined
        # might or might not be set in other cases
        action_point_ids: Optional[List[str]] = None  # list of action point IDs, that are used as action parameters (typically it would be just one, but there might be more)

    data: Data
```

martincervinka, 2022-01-13:
Nicely defined, and described. Maybe just a comment to `action_point_ids` property.

- It is not required for the project without logic. There can be even actions that do not belong to any action point. WebUI will just e.g. highlight all the APs from `action_point_ids` list.
- It will always be defined for the project with logic.

ZdenekM, 2022-01-14:
> It is not required for the project without logic.

Sure, it does not make sense to put there an empty list if the currently executed action does not take any action point as a parameter.

> It will always be defined for the project with logic.

I don't mind... But we will most probably not use it as we already have that information from the project.

closed 2022-01-19 by ZdenekM