robofit/arcor2 #702: Support for runtime Pose parameter

ZdenekM, 2022-01-13:
At the moment, all poses that are used as parameters come from Action Points (pose = point's position + one of its orientations). However, one can't create a program that would e.g. call an action of some object detector returning `Pose` and then use it as a parameter for some consecutive action (e.g. a robot's pick action). We should create a new parameter type, which will, however, be supported **only** for referencing previous results (otherwise it would duplicate the functionality of Action Points).

ZdenekM, 2022-09-02:
It turns out that a new parameter type is not necessary. We can just "link" the pose that is the result of one action to a parameter of another action, which is confirmed to be working (https://github.com/robofit/arcor2/pull/776).

Kapim, 2023-01-13:
Implemented in AREditor.