robofit/arcor2 #749: Align exception format

ChrnyaevEK, 2022-06-20:
Based on earlier discussions error handling will be aligned with the rest of the TIO system. From our side there are plans to update `arcor2_build` and `arcor2_execution_rest_proxy` REST API's. 

## Basic idea of code refactor
This refactor will be applied to both services in the same branch (pull request)

- Ensure all endpoints follow API convention (update paths, query parameters, path variables, responses...)
- Update `arcor2.flask.FlaskException` class to satisfy new definition of WebAPI error (schema based class). Update exception to JSON conversion.
- Ensure that all exceptions are `raised` from original code (not returned as JSON) and then converted to JSON. Update error content and logic to follow new convention.
- Update tests if required (*are there any API specific tests?*)

ChrnyaevEK, 2022-06-27:
Change of `FlaskException` and WebApi interface leads to the need to update all Flask API's. API update is 1:1 rewrite and does not require any discussion so could be done by us  (e.g. dobot, fanuc...)

ChrnyaevEK, 2022-06-29:
I've created a [draft pull request](https://github.com/robofit/arcor2/pull/750) with implemented changes. **Could you please tak a look at it, if such approach is acceptable?** After that I will work with tests and prepare production pull request. 

> Do you mean that the exception itself should become a schema-based dataclass? That should be possible and is quite elegant. Alternatively, FlaskException might take WebApiError dataclass as an argument.

Generally yes, but `FlaskException` is not schema itself, but could be converted to JSON using registered schema. It turned out, that `WebApiError.type` schema must define all error types, that could be raised by service, so we need to generate this data at some point. To solve this issue I implemented the following workflow for services:
1. Service import `FlaskException` and define subclasses based on `FlaskException` overriding `FlaskException.service`, `FlaskException.description` fields.
2. Service request it's own `WebApiError` schema from `WebApiErrorFactory` with all defined exceptions (at this point `WebApiError.type` is generated with desired description).
3. Service pass new `WebApiError` schema to swagger generator (all defined exceptions are shown it `WebApiError` schema description)

Whenever exception based on `FlaskException` is raised by endpoint it is converted to JSON based on registered schema. On the other side it is converted to the "default" `WebApiError` object (defined in `arcor2.data.common.WebApiError`, has no fields descriptions, formally act like abstract class for classes built by `WebApiErrorFactory`) and then this object is passed to `WebApiException.web_api_error`

[Example for build service](https://github.com/ChrnyaevEK/arcor2/blob/align-api-conventions/src/python/arcor2_build/source/exceptions.py)

I've also changed some endpoints in mock services (see #750 - src/python/arcor2_mocks/scripts/mock_scene.py/put_box for example) so it matches REST API and our API conventions (e.g. `put_box` refer box as resource so it's id should be a part oth the path, not query argument). **Is it ok for you?**

ZdenekM, 2022-06-29:
Thanks for the clarification, I think the approach seems fine. I will go through PR and leave there just some minor comments I guess.

ChrnyaevEK, 2022-07-01:
It would be nice to have some `alpha`/`RC` version (preferably today) in case production release will take too long. Is it possible to do so or tests are required for any tag/release version?

ZdenekM, 2022-07-01:
> Thanks for comments, I've commented back and updated PR accordingly. Our team had also have some comments on the subject, so PL include changes requested by our team as well (they are generally about descriptions texts, names of objects/endpoints/parameters etc...).

Ok, I will take a look.

> I'd like to ask, how do you update services versions? After release we will need to have arcor2_rest... in version 0.17.0 and arcor2_build in 0.23.0 (including versions in swagger). Thanks

Well, in the case of `arcor2_execution_rest_proxy`, we used to have the package version and API version separated (a bug fix in code might not have any influence on the HTTP API) - but in this case, we do not use the service so we do not mind :-) Feel free to update `VERSION` files and changelogs so it is prepared for release.

> It would be nice to have some alpha/RC version (preferably today) in case production release will take too long. Is it possible to do so or tests are required for any tag/release version?

Why not, I can make a pre-release. However, without checks and tests passing, we can't merge into the `master` branch. I guess that it should not be a big deal to get tests to succeed. First, please format sources and tune type-checking errors. Then tests will attempt to run and we will see. Releasing something without the tests passing does not make big sense to me, as it might be broken and unusable.

ChrnyaevEK, 2022-07-01:
Ok, it's clear now. So, may you please make pre-release tag for `arcor2_build` and `arcor2_execution_rest_proxy`. I will continue working on PR and will inform you as soon as it is ready. Thanks

ZdenekM, 2022-07-01:
Sorry, I wasn't very clear probably :) I would prefer to first make tests pass, then make a pre-release, then (if needed) fix bugs in another PR, and finally do an ordinary release. So far, we always did it like this - first merge the stuff into `master` and then release. As I wrote above - releasing something without the tests passing does not make big sense to me, as it might be broken and unusable. The thing is also that the release process is not fully automated so far, so it is not effortless.

ZdenekM, 2022-07-04:
Yes, that's correct, we also update changelogs manually.

ChrnyaevEK, 2022-07-07:
Finally #750 PR is ready for review. I've changed `WebApiError` to be JSON based exception and removed `RestHttpException`  as HTTP error response code does no longer make senese (when error happens 500 is returned in any case). So on the one side service raise `WebApiError` and it gets converted to JSON and on the other side `arcor2.rest` covert HTTP500 JSON body to `WebApiError` object and raise it.

`arcor2_kinect_azure` and `arcor2_calibration` were updated as well  (I've missed them earlier by mistake). 

In the future we will have to update `arcor2.clients.scene_service , project_service` and remove @handle wrapper (it does not allow `WebApiException` to be raised properly)

So, I guess this are the final changes and as soon as you approve PL new `arcor2_build` and `arcor2_execution_rest` may be released.

ZdenekM, 2022-07-08:
Ok, thank you for all the changes (especially in our own services). Build and Execution Proxy services are released.

closed 2022-07-08 by ZdenekM