robofit/arcor2 #762: Switch to Python 3.10

ZdenekM, 2022-08-02:
To keep the system up to date, for better performance, and to be able to use some newer language features, the repo should switch to Python 3.10. Currently, this is blocked by `open3d`, which is only for Python 3.9. However, its next release should also support 3.10. See https://github.com/isl-org/Open3D/issues/4535 -- once this is closed, let's try to switch.

ZdenekM, 2022-08-02:
@ChrnyaevEK Is switching to 3.10 ok for Kinali? It won't be soon probably...

ChrnyaevEK, 2022-08-02:
It sounds very interesting and I guess it will not be such a big deal for us to switch. Terms are also ok as we do not plan any significant changes that would affect `arcor2` yet. Thanks for early notification!

ZdenekM, 2022-10-28:
Done... All packages were released.

closed 2022-10-28 by ZdenekM